jefmenegazzo/mpu-i2c-drivers-python #4: Wide Range in Values from Magnetometer

rdeliaus, 2020-02-14:
I am calibrating the sensors with the functions. The Accelerometer and Gyroscope readings appear to be fine, however the readings I am getting from the Magnetometer are showing a wide range of values.

I have the MPU-9250 mounted with two motors/wheels in a robot on a raised platform (so it doesn't move during testing). I calibrate the Magnetometer which reads near zero values. Upon starting and stopping the motors (again no real movement), the Magnetometer values swing pretty wildly (-10 + 20).  What is really strange is when the motors are stopped, the Magnetometer values never goes back to near zero but it keeps these high values.

I also isolated the 3.3v power to a clean source to make sure noise was not a factor.

I suspect this is either a problem with the Magnetometer calibration function, or the physical layout / circuit/device. Any thoughts are much appreciated.

Thanks.

Raspberry Pi Zero W
SparkFun IMU MPU-9250 using I2C communication

jefmenegazzo, 2020-02-22:
@rdeliaus 

**Some considerations:**

The calibration function adopted in this repository is widely used, being one of the most common, due to its simplicity and effectiveness. It corrects both soft iron distortion and hard iron distortion. 

The calibration of the magnetometer was programmed based on this widely used repository:
- https://github.com/kriswiner/MPU9250/

The file containing the calibration code is this:
- https://github.com/kriswiner/MPU9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

Here you can find out how this calibration works:
- https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration

And here you can see how to test the calibration:
- https://appelsiini.net/2018/calibrate-magnetometer/

Since this is one of several possible forms of calibration, another more sophisticated one is found here (in Python too):
- https://teslabs.com/articles/magnetometer-calibration/

**My sugestion:**

- Check if there is an issue in kriswiner repository similar to your problem, checking if what happened is an expected behavior or not.

- Clone my repository and change the magnetometer calibration function to that of teslabs.

Soon I will make tests with this repository code and my hardware to give you a more assertive answer. I ask you to keep this issue up to date with what you discover.

bevinhex, 2020-03-01:
I'm using sparkfun 9250 red sensor, calibration for accelrometer , gyro, megnetometer all seems fine, getting nice data after complementary filter fusion, using http://wiki.ros.org/imu_complementary_filter?distro=melodic, I was about to write my own library then saw this one, thank your for the great work done ! I'll post or help out any issues came up with, since I do have a couple of versions of 9250 around as well.

bevinhex, 2020-03-01:
[image]
so far mine swings around +-6 uT, using 16bit resolution, think it is pretty good comparing what I got another sensor, perhaps you can install an app on mobile to see its sensor data and compare it with the one you get from your sensor

bevinhex, 2020-03-01:
that is (uT) showing there, because I forgot to turn them into Tesla

jefmenegazzo, 2020-06-16:
Closed due to inactivity.

closed 2020-06-16 by jefmenegazzo