jefmenegazzo/mpu-i2c-drivers-python #6: Thanks, SMBUS2 and data management

bbelmar, 2020-04-16:
First thank you for this amazing work!

I had to dig into the other files to realize I needed to Install smbus2. I had smbus and it was automatically using fake smbus. Maybe user error since im a newbie.\

Is there a way to get each measurement data as a separate value? for example:
acceleration in X = ##
acceleration in Y = ##
acceleration in Z = ##

gyro X = ##
gyro Y = ##
etc.
Thank you very much!

jefmenegazzo, 2020-04-16:
@bbelmar Are you using the library through `git clone` or `pip install`?

If you used  `git clone`, you executed the `pip install -e .` ? This command should install smbus2. 

As you can imagine, smbus and smbus2 are two different libraries. In this project only smbus2 is used because it is faster and more consistent than smbus (I did several tests).

Regarding the use of each axis separately, there is no implementation to read one axis at a time from the sensor, since reading each separately has a worse performance than reading them all at once.

If you just want to use them separately, you could read them all and assign each axis variable like this:

```python
data= mpu.readAccelerometerMaster()
accelerationX = data[0]
accelerationY = data[1]
accelerationZ = data[2]
```
Does it help you?

bbelmar, 2020-04-16:
Meu Deus cara, vc e incrivel. Muito obrigado por responder rapido!
Vou tentar agora aquilo que vc sugeriu. Brigado!

bbelmar, 2020-04-16:
That worked perfectly to show the data separately. But Accel and Gyro data for some reason resets itself to a range of -.5 and =.5 approx.
If i move the sensor or rotate in any axis i see the data show up but as the sensor remains stationary in its new position, the readings go back to -.5 to =.5, thats not supposed to happen right?

robert-meiner, 2020-06-04:
> @bbelmar Are you using the library through `git clone` or `pip install`?
>
> If you used `git clone`, you executed the `pip install -e .` ? This command should install smbus2.
>
> As you can imagine, [...]

I used the 'pip install -e .' method and it did not install the smbus2 library.  At first, I thought my sensor wasn't working, so I tried it on an Arduino, where it works fine. 

fix this with:

sudo python3 -mpip install smbus2

Everything works now.

Thanks!

jefmenegazzo, 2020-06-16:
@bbelmar Check the library's calibration features.

@robert-meiner added command to install dependencies in the documentation.

`pip install -r requirements.txt`

closed 2020-06-16 by jefmenegazzo