jefmenegazzo/mpu-i2c-drivers-python #7: [Errno 16] Device or resource busy

reneplatypus, 2020-06-24:
**Describe the bug**
When I run:
`python3 master-only-mode.py`

I get:

> Traceback (most recent call last):
>   File "/usr/local/lib/python3.6/site-packages/mpu9250_jmdev/mpu_9250.py", line 95, in configure
>     self.configureMPU6500(self.gfs, self.afs)
>   File "/usr/local/lib/python3.6/site-packages/mpu9250_jmdev/mpu_9250.py", line 135, in configureMPU6500
>     self.writeMaster(PWR_MGMT_1, 0x00, 0.1)
>   File "/usr/local/lib/python3.6/site-packages/mpu9250_jmdev/mpu_9250.py", line 1022, in writeMaster
>     self.bus.write_byte_data(self.address_mpu_master, register, value)
>   File "/usr/local/lib/python3.6/site-packages/smbus2/smbus2.py", line 429, in write_byte_data
>     self._set_address(i2c_addr, force=force)
>   File "/usr/local/lib/python3.6/site-packages/smbus2/smbus2.py", line 333, in _set_address
>     ioctl(self.fd, I2C_SLAVE, address)
> OSError: [Errno 16] Device or resource busy

(3 times)

**To Reproduce**

Enabled I2C using:
`echo -e "dtparam=i2c_arm=on\ndtoverlay=i2c-rtc,ds3231" >> /boot/config.txt`
`echo -e "i2c-dev" >> /etc/modules-load.d/i2c.conf`
`yum -y install i2c-tools && reboot`

Then I enabled the SMBus2 by using from MPU9250 folder:

`pip3 install -r requirements.txt`
 
Then ran:

`python3 master-only-mode.py`

**Expected behavior**
I expected to start reading values from address 68.
**Desktop:**
 Running Centos 7 on a Raspberry Pi 3 B+.
I have both python 2 and python3 so I need to be using pip3 and python3.

jefmenegazzo, 2020-06-25:
@reneplatypus 

Does every time you run the code result in this problem? Or was any execution successful?

You can use the i2c-tools commands to test if your hardware has any problem.

Run:

`i2cdetect -y 1`

And check if your MPU is showing at address 0x68. If it's ok, run the command:

`i2cget -y 1 0x68 0x3B`

If the same problem occurs, it may be some configuration of the sensor or the hardware. You can manipulate registers via command line with i2cget and i2cset to test (https://www.mankier.com/package/i2c-tools).

Another test you can do is in the mpu_9250.py class change:

```python
# import smbus
 import smbus2 as smbus
```

To:

```python
import smbus
# import smbus2 as smbus
```

And rerun.

jefmenegazzo, 2020-07-04:
@reneplatypus Any updates on that?

jefmenegazzo, 2020-07-22:
Closed due to inactivity. Probably a hardware problem.

closed 2020-07-22 by jefmenegazzo

reneplatypus, 2020-07-22:
Sorry.  Thank you for the follow up. I have been busy on other projects and will soon retake this one using the MPU-9255.  I have it on my desk to remind me to get back to it.

As soon as I do, I will try your suggestion and will post.

Thank you again.

René Patterson
rene@platypusinc.net
Cel: 6613-9188