jefmenegazzo/mpu-i2c-drivers-python #8: TypeError: configureAK8963() missing 2 required positional arguments: 'mfs' and 'mode'

robert-meiner, 2020-07-17:
Dear Colleagues,  

I am getting the following error, when attempting to write calibration values to the AK8963 in the Invensense MPU9250 module.  I am able to calibrate the accelerometer and gyro.

```Traceback (most recent call last):
  File "mpu9250_cal.py", line 232, in <module>
    main()
  File "mpu9250_cal.py", line 218, in main
    calmag(mpu)
  File "mpu9250_cal.py", line 142, in calmag
    mpu.configureAK8963() #resets the calibration values of the AK8963
TypeError: configureAK8963() missing 2 required positional arguments: 'mfs' and 'mode'
```
Here are the libraries I've imported:
```  1 import sys
  2 import math
  3 import time
  4 from smbus2 import SMBus
  5 from mpu9250_jmdev.registers import *
  6 from mpu9250_jmdev.mpu_9250 import MPU9250```

Here is the magnetometer calibration function in my test code-- when the error appeared, i created these clunky variables and got rid of a lot of other things to try to troubleshoot:

```99 #normalize hard and soft iron biases of magnetometer -- robot needs to be waved in the air in figure 8 patterns for 30 sec.
100 def calmag (mpu):
101     global masterMag
102     global mbias
103     global magScale
104     magMax = [0,0,0] #placeholder for mag data
105     magMin = [0,0,0] #placeholder for min mag data
106     mbiasn = 0 #hard iron x bias value to be written to AK8963
107     mbiase = 0 #hard iron y bias value to be written to AK8963
108     mbiasd = 0 #hard iron z bias value to be written to AK8963
109     mscalen = 0 #soft iron x estimate to be written to AK8963
110     mscalee = 0 #soft iron y estimate to be written to AK8963
111     mscaled = 0 #soft iron z estimate to b written to AK8963
112     
113     #write values to AK8963 magnetometer
114     mpu.configureAK8963() #resets the calibration values of the AK8963
115     mpu.configure()
116     mpu.bias = [mbiasn, mbiase, mbiasd] #write estimated hard iron biases
117     mpu.magScale = [mscalen, mscalee, mscaled] #write estimated soft iron biases
118     
119     #print values
120     print("hard iron values ",mpu.mbias)
121     print("soft iron values ",mpu.magScale)
122     print("\n")
123     return```

and here's where I call the function from main()

```189     input("When you have the robot in your hand and are ready, press the return key")
190     calmag(mpu)```

Running on headless raspberry pi 4b+ using debian buster

I'd appreciate your thougts on how to resolve this.  I have't been able to find information on the positional arguments in the error message.

Thanks, Mark

jefmenegazzo, 2020-07-17:
@robert-meiner Check if in the construction of the MPU9250 object you are passing the parameters commented below:

```python
mpu = MPU9250(
    address_ak=AK8963_ADDRESS, 
    address_mpu_master=MPU9050_ADDRESS_68,
    address_mpu_slave=None, 
    bus=1,
    gfs=GFS_1000, 
    afs=AFS_8G, # Check this
    mfs=AK8963_BIT_16, # Check this
    mode=AK8963_MODE_C100HZ
)
```

robert-meiner, 2020-07-17:
I just found the problem.  In the instructions on the code page and on the [project pypi.org page] (https://pypi.org/project/mpu9250-jmdev/), the instruction to calibrate the AK8963 is given incorrectly.  

The mistake is located under:

**Magnetometer**

To perform calibration run the command:
```    mpu.configureAK8963() # Calibrate sensors
    mpu.configure() # The calibration function resets the sensors, so you need to reconfigure them

    magScale = mpu.magScale # Get magnetometer soft iron distortion
    mbias = mpu.mbias # Get magnetometer hard iron distortion
```

The Command should be ```mpu.calibrateAK8963()```

An easy mistake to make, when writing such detailed instructions.  My code works now.

closed 2020-07-17 by robert-meiner