jefmenegazzo/mpu-i2c-drivers-python #9: Cannot calibrate slave

jessicatrac, 2020-09-03:
Hello! Thanks so much for your work - greatly appreciated. I'm working with two MPU9250s for a simulator project and am having trouble getting data from the slave consistently. When I start up my Pi for the first time and run the master-slave-mode.py example without the calibration, I get reliable readings for the master and slave. For example, accelerometer readings of [0,0,1] on flat ground and [0,0,-1] when flipped upside down. Gyroscope readings are close to [0,0,0] while stationary for the most part.

When I run mpu.calibrate() after, I run into this error:

> Calibrating 0x68 - AK8963
> Traceback (most recent call last):
>   File "/home/pi/MPU9250/examples/basic-usage/master-slave-mode.py", line 36, in <module>
>     mpu.calibrate() # Calibrate sensors
>   File "/home/pi/MPU9250/mpu9250_jmdev/mpu_9250.py", line 592, in calibrate
>     self.calibrateAK8963()
>   File "/home/pi/MPU9250/mpu9250_jmdev/mpu_9250.py", line 887, in calibrateAK8963
>     avg_rad / mag_scale[0],
> ZeroDivisionError: float division by zero

Since I am most concerned with getting accel & gyro readings for my project, I switch to mpu.calibrate6500(), but run into this error next:

> Could not connect to slave to calibrate
> |.....MPU9250 in 0x68 Biases.....|
> Accelerometer Master [0.12724958147321427, 0.08103143601190477, -0.13961937313988093]
> Accelerometer Slave [0, 0, 0]
> Gyroscope Master [-2.0375932965959818, -0.30699230375744047, -0.20599365234375]
> Gyroscope Slave [0, 0, 0]
> Magnetometer SID [1, 1, 1]
> Magnetometer HID [0, 0, 0]

At first I thought it might be a wiring problem with the slave IMU. However, since the initial readings look fine I am not sure if that is the case. Any guidance would be appreciated!

jefmenegazzo, 2020-10-08:
@jessicatrac Could you give the following information?

- Are you using the pypi.org library `mpu9250-jmdev` or clone this repository?
- Which SBC are you using, Raspberry PI?
- What type of cables are you using? Jumpers like [these](https://produto.mercadolivre.com.br/MLB-807651191-kit-40-jumpers-macho-fmea-30cm-cabo-fios-protoboard-jumper-_JM?matt_tool=79246729&matt_word=&gclid=CjwKCAjw_NX7BRA1EiwA2dpg0tRNluGppfOiKHAUhEqGcnKlNDHT-bZz0hL9t_TEmeaOPVawf90P3xoCYpYQAvD_BwE)? Or Manga like [these](https://produto.mercadolivre.com.br/MLB-1133000519-25-metros-cabo-manga-6-vias-26awg-preto-sem-blindagem-_JM?matt_tool=51680284&matt_word&gclid=Cj0KCQjwpPHoBRC3ARIsALfx-_LlW450xpzGrSgD8dY2IYH8udzTaxE0j20bboNo2J61fQUEuphtzp8aAs12EALw_wcB&quantity=1)?
- How long are the cables?

jessicatrac, 2020-10-21:
@jefmenegazzo sure! Thanks for the help.

- Are you using the pypi.org library mpu9250-jmdev or clone this repository? **I was previously using the pypi.org library but changed it over to the git repo clone. I am still running into the same problem EXPECT sometimes I am able to calibrate the slave - but once it hits the 'cannot connect slave to calibrate', it stays like that until reboot.** 

[oct21_mpu9250_jmdev_test.txt](https://github.com/Intelligent-Vehicle-Perception/MPU-9250-Sensors-Data-Collect/files/5418734/oct21_mpu9250_jmdev_test.txt) -> here's some logs from a test I did. 2 successful data collections prior to any calibration. 2 successful calibrations next (although the 2nd doesn't seem as valid). Then 3 unsuccessful calibrations.

- Are you using the latest version of the library/code? **Yes**
- Which SBC are you using, Raspberry PI? **Yes, Raspberry Pi 4.**
- What type of cables are you using? Jumpers like these? Or Manga like these?  **I am using ribbon cables. I am attaching a photo of my set up.**
- How long are the cables? **50cm. I thought this may be the issue but if you're able to go up to 2.5m maybe that isn't the case.**

[image]

- Have you tested the slave MPU as a master to verify that there is no hardware problem? **Yes, both work separately as a master.**

I am only using two MPU9250s and have them wired up as your diagram is showing for MPU0x68 Master and Slave. I don't suspect it's a wiring issue since the solder points are fine and length is as well. I may try the 0x69 route for one but I recall having some trouble with it before. I also think it might be signal loss due to a hardware error but cannot confirm. My breakout board for the MPU9250 is also different from the one you linked on pypi, which is likely not a factor. I'm using [Sparkfun](https://www.robotshop.com/en/imu-breakout-board-mpu-9250.html). I think I'll attempt the 0x69 route again and then order some new sensors in case it's a true hardware issue. 

I will give you another update once I try a couple of the above. Please let me know if any of my answers above give you any further insight into my problem!

jessicatrac, 2020-12-16:
@jefmenegazzo thanks for your help. In the end I had set the MPU addresses to 0x68 and 0x69 and create two separate objects. I was not able to get the data collection working by assigning the addresses as the master and slave. Not sure why, as they work separately and hardware seems fine!

closed 2021-01-16 by jefmenegazzo