jefmenegazzo/mpu-i2c-drivers-python #12: no values for magnometer,

pdrewes, 2021-03-10:
only x68 shows up, 
 acceleration / velocity work fine running master-slave-mode.py 
 raspberry pi 3b, latest OS (Jan 2021) just loaded ran the install scripts as desired from the instructions.

running master-only-mode.py
gets
 File "/usr/local/lib/python3.7/dist-packages/mpu9250_jmdev/mpu_9250.py", line 1012, in writeAK
    self.bus.write_byte_data(self.address_ak, register, value)
  File "/usr/local/lib/python3.7/dist-packages/smbus2/smbus2.py", line 434, in write_byte_data
    ioctl(self.fd, I2C_SMBUS, msg)
OSError: [Errno 121] Remote I/O error
thread sampling - run.py gets the same I/O error.
VCC 5v, GND, , SCL / SDA connected

jefmenegazzo, 2021-03-14:
@pdrewes 

Please provide more information about your environment:

- Which platform? Raspberry Pi?
- Which MPU model? 9250?
- How are you using it? Just an MPU? Two, a slave and a master?
- Provide your source code.

If you are using something simple like just an MPU in RPi, use the **master-only-mode.py**' file in the examples folder.

pdrewes, 2021-03-14:
Just raspberry pi 3b into 9250 nothing else connected 

Your source code the master only or the multi gets the same problem. 
Only x68 shows up. 

I get the acceleration but no compass readings. Device is good.

jefmenegazzo, 2021-03-15:
- The _OSError: [Errno 121] Remote I/O error_ means that the OS's I2C bus could not find the address of your magnetometer (AK8963). So it's a hardware problem, not this library.

- Run `sudo i2cdetect -y 1`, your magnetometer should appear at address 0x0C.

- If you are not using the MPU as a master slave, then use only **master-only-mode.py** as described in the documentation.

- I recommend running:

```python
import sys
sys.path.append("")

import time
from mpu9250_jmdev.registers import *
from mpu9250_jmdev.mpu_9250 import MPU9250

mpu = MPU9250(
    address_ak=AK8963_ADDRESS, 
    address_mpu_master=MPU9050_ADDRESS_68, # In 0x68 Address
    address_mpu_slave=None, 
    bus=1, 
    gfs=GFS_1000, 
    afs=AFS_8G, 
    mfs=AK8963_BIT_16, 
    mode=AK8963_MODE_C100HZ)

mpu.reset() 
```

Afterwards, restart your RPi and re-run `sudo i2cdetect -y 1` to see if the address 0x0C appears.

closed 2021-03-15 by jefmenegazzo

pdrewes, 2021-03-15:
But using other applications I can see both. Through just the 68 port. 

I was hoping to use this library 

Not hardware fault.

JGonzalga, 2021-06-21:
Hello, I am struggling with the same problem, I only see the 0x68 device when running sudo i2cdetect -y 1 and then the master-only-mode.py script shows the same error: OSError: [Errno 121] Remote I/O error

were you able to solve it? 

Thanks,
Josué

pdrewes, 2021-06-21:
I was not able to solve it with these libraries. I had to use other ones to get all the data. 
It would not recognize the magno  portion of the sensor no matter what I did.

JGonzalga, 2021-06-21:
> I was not able to solve it with these libraries. I had to use other ones to get all the data. It would not recognize the magno portion of the sensor no matter what I did.
> […](#)

Can you share which libraries did work for you?