jefmenegazzo/mpu-i2c-drivers-python #15: Getting Errno 121: remote I/O error

pkoprov, 2021-09-27:
I installed the library without smbus2 at first and it was running fake smbus correctly. After I installed smbus2 the example code stopped working. it gives an error 121. what should I do?

jdc-cunningham, 2022-01-21:
Yeah sucks about the hit/miss of sensor quality.

JafetSalgado, 2022-11-13:
I have exactly the same issue
[image]
[image]

campagnani, 2022-11-13:
@JafetSalgado the magnetometer address is 0x0C and don't appaers in the image of terminal you send here.

Maybe you burn it pluggin the sensor in 5v?

I did the same thing (plug the MPU9250 in 5v) and burn just the IC of magnetometer.

closed 2023-01-06 by pkoprov